sm6yvr/liam #23: Remove API_RESPONSE enum and create a new API_STATUS_CODE

sm6yvr, 2017-09-05:
Create a new API_STATUS_CODE enum with OK=1, ERROR=2, ARGUMENT_ERROR=3 and use this as the second field in a response message

Ola-Palm, 2017-09-06:
Ok såhär?

enum API_STATUS_CODE {
  OK=1,
  ERROR=2,
  ARGUMENT_ERROR=3,
  HEARTBEAT=4,
 (  SUBSCRIBE=10, //  ) Ej klart om behövs ännu.. 
  DEBUG=90,
  ONLINE=99
};

sm6yvr, 2017-09-06:
Ska HEARTBEAT också vara en statuskod? Vi får väl ha en HEARBEAT i API_COMMAND-enumen men då räcker det väl att skicka OK som status på HEARTBEAT?

Ola-Palm, 2017-09-06:
Jag tycker att heartbeat är en status på om linjen mellan xx och Arduinon lever. Får man inte ;4# efter utsatt tid för heartbeat så kan man inte längre lita på att informationen som man visar på GUI. 
I Fråga svar så faller ju sig detta per automatik, men om vi ändrar detta längre fram till att inte fråga efter status, utan blir uppdaterade när en statusförändring sker, så kan de vara skönt att se ett räkneverk som går efter heartbeat.

sm6yvr, 2017-09-06:
Jag håller med, men då ska inte HEARTBEAT vara en status utan ett command. vi sa ju att en range (0-99?) skulle vara sånt som aruinon kan skicka utan att klienten har skickat en request. Det viktiga är nog att hålla rätt format i det som skickas från arduinon ";<command>:<status>#".

Ola-Palm, 2017-09-06:
mmm. Jag tycker att ett hjärtslag är en status på att saker lever + att det kommer utan att man ber om det. ;4# skickas ut på serie snöre så fort Arduino är igång utefter det intervall som vi sätter.  så jag tycker absolut att det faller in i range 0-99. Men jag kan absolut ändra om du tycker det. 
Vad ska det i så fall vara?

sm6yvr, 2018-05-10:
Closing the issue because it's not applicable anymore..

closed 2018-05-10 by sm6yvr